Muslinmin/glowing-waffle #1: Familiarization with version control

Muslinmin, 2024-01-06:
**GitHub Basics**

GitHub is where all our project files are stored and is what we use for version control.

- [] Learn how to assign issues by assigning yourself to this issue using the Assignees row in the column on the right of this comment
- [] Learn how to organize issues by adding a "version_chores" label to this issue using the Labels row in the column on the right of this comment
- [] Explore the Issues tab to understand how you may browse and view issues
- [] Explore the Wiki tab to understand how we can use GitHub's Wiki to document things

**GitHub for Source Control & Development**

 GitHub is also used as a project management stand-in app, where we post our issues (tracks), deadlines, etc.

- [] Create your own dev-branch e.g. dev-reuben and reference from the main branch
- [] Push some random updates to your own branch
- [] Showcase your repository by commenting here with a link, https://github.com/LordMuze/cli, using the edit mode on the top right of this issue. 
- [] (Optional) Understand how to use GitHub Desktop

- [] Add this issue to the Kanban board under the Projects tab --> Projects --> Embedded Systems Project
- [] Continue adding a Status, Priority, Effort and Deadline to this issue using the rows on the right of this page
- [] Explore how the Kanban board works by navigating it to from the top menu, Projects --> Projects --> Immersification Kanban
- [] Learn how to create issues by converting the list 'to be converted' task-list to issues.

- [ ] 'to be converted'
- [ ] 'to be converted'

Muslinmin, 2024-01-13:
So are you still keen on using this tool? @s2200931

s2200931, 2024-01-13:
> So are you still keen on using this tool? @s2200931

Yes, your grace.

closed 2024-01-15 by s2200931