Muslinmin/glowing-waffle #17: Lab 4

Muslinmin, 2024-01-25:


Muslinmin, 2024-01-27:
For now we just add the main scripting file and a zip file containing the .ioc to the branches. So I am envisioning dev-mus2, dev-reub2 and proj2 branches @s2200931

Muslinmin, 2024-01-29:
- https://www.embedded.com/interrupts-short-and-simple-part-1-good-programming-practices/

There is an unwritten rule not mentioned by Ian but I observed his code and he generally does this:

Function calls and normal CPU routines should be handled in the while loop and not the interrupt routine
- Routine calls should be reserved for state machine configuration i.e. setting of flags only

Essentially the routine calls should not be too long as it may lead to CPU undefined state and some horrible (hard to define) runtime errors

The website link above gives tips and an ideal approach to interrupt driven programming

@s2200931 @Chillhopper 

So now I shall just focus on refactoring the previous Lab3 code. Anyways the current baseline is not so friendly when it comes to adding more features - A state machine (structs) approach makes the code scalable

Muslinmin, 2024-01-30:
@s2200931 This is the small reference of how it was done inspired by Professor's Ian and something which we would use as
an architecture in the future.

Below just contains the .c file. No changes made to .ioc

[lab3_uart2only.zip](https://github.com/Muslinmin/glowing-waffle/files/14092909/lab3_uart2only.zip)

s2200931, 2024-02-01:
Output on HTERM when AT+RESTORE command is sent:
```
<\0><\0>AT+RESTORE<\r><\n>
<\0><\0><\r><\n>
OK<\r><\n>
<\0><\0>tail 0<\r><\n>
chksum 0xf3<\r><\n>
load 0x3ffe8000, len 764, room 8 <\r><\n>
0x40100000, len 2592, room 16 <\r><\n>
<\0><\0>tm 0x92<\r><\n>
load 0x3ffe82fc, len 676, room 4 <\r><\n>
tail 0<\r><\n>
chksum 0x22<\r><\n>
csum 0x22<\r><\n>
<\r><\n>
2nd boot ver<\0><\0>òŒ<\f><\0>
```

Error output on console:
```
Retrying... UART1 HAL_UARTEx_ReceiveToIdle_DMA 
 UART1 HAL_UARTEx_ReceiveToIdle_DMA cannot start. 
```

When the AT+RESTORE command is transmitted to the ESP chip, this causes the chip to go inactive for a moment which could cause issues that leads the MCU to think that UART1 has gone into IDLE and prepares to receive data when in reality there is no data to be received. This is one of the error cases Prof Ian has mentioned.

Muslinmin, 2024-02-01:
> Output on HTERM when AT+RESTORE command is sent:
>
> ```
> <\0><\0>AT+RESTORE<\r><\n>
> <\0><\0><\r><\n>
> OK<\r><\n>
> <\0><\0>tail 0<\r><\n>
> chksum 0xf3<\r><\n>
> load 0x3ffe8000, len 764, room 8 <\r><\n> [...]

Apparently for my Wifi chip nothing was sent from the ESP01 chip...

Muslinmin, 2024-02-01:
Anyways, I found a resource which I think it give me a slight hint but not so helpful in implementing what exactly to do because its pretty technical for me to understand at current capability now @s2200931 

https://www.youtube.com/watch?v=8-y1bRr1Qis

Muslinmin, 2024-02-01:
```
		if(uart1.rxState == UART_ERROR){
			printf("Retrying... UART%s HAL_UARTEx_ReceiveToIdle_DMA \r\n ", (char*) (uart1.Text));
			rx_UART_Receive_ErrorRecovery_Routine(&uart1);
			onIdle_UARTReceive(&uart1);
		}
```

```
void rx_UART_Receive_ErrorRecovery_Routine(uartPropertyType* uart){
	  if(HAL_UART_DeInit(uart->h_UART) != HAL_OK)
	  {
		  printf("De-initialized UART%s ERROR! \r\n", uart->Text);
//	    Error_Handler();
	  }else{
		  printf("De-initialized UART%s \r\n", uart->Text);
	  }
	  if(HAL_UART_Init(uart->h_UART) != HAL_OK)
	  {
		  printf("Re-initialized UART%s ERROR! \r\n", uart->Text);
//	    Error_Handler();
	  }else{
		  printf("Re-initialized UART%s \r\n", uart->Text);
	  }
}
```

As mentioned in the video, the one of the errors requires re-initialization of the peripheral. @s2200931 
Now we have to handle the remaining errors:

1. Framing Error
2. Parity Error
3. Overflow/Overrun

@s2200931

Muslinmin, 2024-02-01:
I kept spamming

[image]