Muslinmin/glowing-waffle #18: String Operations Overflow

Muslinmin, 2024-01-30:
```
void on_ReceiveFrom_UART2_To_UART1(uartPropertyType* uart){
    //debugText + currentBuffer
	char* debugText;
	size_t debugTextLength;

	debugText = "UART1_Rxed: ";
	debugTextLength = strlen(debugText);
	size_t messageLength = strlen(RxBuffer) + debugTextLength + 1; // to place the null character

	if (debugTextLength + messageLength < RX_BUFSIZE) {
		//1 - MOD msg
		memmove(RxBuffer + debugTextLength, RxBuffer, strlen(RxBuffer) + 1); // +1 for null terminator
		memcpy(RxBuffer, debugText, debugTextLength);
		printf("In Modify Message: %s \r\n", RxBuffer);
	}else{
		printf("on_ReceiveFrom_UART2_To_UART1() : Not enough space!\r\n");
	}
}

void on_ReceiveFrom_HTERM_To_UART2(uartPropertyType* uart){
	    //debugText + currentBuffer
		char* debugText;
		size_t debugTextLength;
		size_t messageLength = uart->rxDataSize;
		debugText = "UART2_Rxed: ";
		debugTextLength = strlen(debugText);

		if (debugTextLength + messageLength < RX_BUFSIZE) {
			//1 - MOD msg
			memmove(RxBuffer + debugTextLength, RxBuffer, messageLength+1); // +1 for null terminator
			memcpy(RxBuffer, debugText, debugTextLength);
			printf("In Modify Message: %s \r\n", RxBuffer);
		}else{
			printf("on_ReceiveFrom_HTERM_To_UART2() : Not enough space!\r\n");
		}
}
```
on_ReceiveFrom_HTERM_To_UART2() receives a size of an exact number of characters. 

on_ReceiveFrom_UART2_To_UART1() receives the full size of RxBuffer but -1 for the null character. 

The problem is on the second time I send message from Hterm to Stm32  this portion:

	if (debugTextLength + messageLength < RX_BUFSIZE) {
		//1 - MOD msg
		memmove(RxBuffer + debugTextLength, RxBuffer, strlen(RxBuffer) + 1); // +1 for null terminator
		memcpy(RxBuffer, debugText, debugTextLength);
		printf("In Modify Message: %s \r\n", RxBuffer);
	}else{
		printf("on_ReceiveFrom_UART2_To_UART1() : Not enough space!\r\n");
	}

is buggy because nothing is printed. memmove() and memcpy have some hidden implications that you might wanna avoid and create your own string operations instead...

Regardless, now  

```
 __HAL_DMA_DISABLE_IT(&hdma_usart1_rx, DMA_IT_HT | DMA_IT_TC);
  __HAL_DMA_DISABLE_IT(&hdma_usart2_rx, DMA_IT_HT | DMA_IT_TC);
```
is working as expected thanks to the refactoring shift - letting the interrupts to handle ONLY STATES and that is all

Muslinmin, 2024-01-30:
The problem was that initially, the buffer of 65 bytes(includes the null character at the end) was received by USART1, and then  STM32 decided that the USART 1 Receive Line is ready for incoming data.

Subsequent message, i.e. we go USART2 Receive -> USART 1 SEND

USART1 Send operations is deemed successful because the Transfer Complete callback was invoked. The HAL_UARTEx_ReceiveToIdle_DMA however, fails to receive anything so far, so the HAL_UARTEx_RxEventCallback() is never called for USART1.

Just because the USART1 Send operations is successful does not mean the receiving end is successful. Since interrupts for DMA completed receive and half completed receive is disabled, only the IDLE interrupt for UART is enabled.

And so, here my conclusion is, the Receive Line didnt go IDLE. Why?

As mentioned earlier in this comment, the buffer is not all used up. Some characters only filled a portion of the buffer. The remaining rest are null characters. And yet, I passed the entire buffer size which includes the characters that was filled and a majority of null characters. 

**So, the actual buffer size i.e. only the filled up characters + one null character is passed to the Transmit function.** Im assuming, there was an attempt by the transfer peripheral to transmit null characters over the line, but failed to do so without raising any flag errors, hence the UART line never goes IDLE... @s2200931

s2200931, 2024-01-31:
Noted, that's an awesome find.

s2200931, 2024-01-31:
1. Refactored onIdle_UARTReceive and UART_Transmit_WithState into modMessage.
2. Messages are transmitted inside modMessage now using UART_Transmit_WithState.
3. Issue with byte size transferred remains with uart->txStatus = HAL_UART_Transmit_DMA(uart->h_UART, RxBuffer,  uart->txDataSize); where byte size transferred is insufficient causing transmission failure. 

`void onIdle_UARTReceive(uartPropertyType* uart){
//	uart->rxStatus = HAL_UARTEx_ReceiveToIdle_DMA(uart->h_UART, RxBuffer, (RX_BUFSIZE - 1));
	uart->rxStatus = HAL_UARTEx_ReceiveToIdle_DMA(uart->h_UART, RxBuffer, RX_BUFSIZE);
	if(uart->rxStatus == HAL_OK){
		__HAL_DMA_DISABLE_IT(uart->hDmaRx, DMA_IT_HT | DMA_IT_TC);
		uart->rxState = RECEIVING;
	}else{
		uart->rxState = UART_ERROR;
		printf("UART%s HAL_UARTEx_ReceiveToIdle_DMA cannot start. \r\n", (char*) (uart->Text));
	}
}

void UART_Transmit_WithState(uartPropertyType* uart){
	__HAL_DMA_DISABLE_IT(uart->hDmaTx, DMA_IT_HT);
	uart->txStatus = HAL_UART_Transmit_DMA(uart->h_UART, RxBuffer,  uart->txDataSize);
//	uart->txStatus = HAL_UART_Transmit_DMA(uart->h_UART, RxBuffer,  uart->txDataSize);
	if(uart->txStatus == HAL_OK){
		uart->txState = TRANSMITING;
	}else{
		uart->txState = UART_ERROR;
		printf("UART%s HAL_UART_Transmit_DMA cannot start. \r\n, ", (char*) (uart->Text));
	}
}

void modMessage(uartPropertyType* uart) {
	char* debugText;
	size_t debugTextLength;
	size_t messageLength = strlen((char*)RxBuffer);

	switch(uart->uartID) {
		case 1:
			debugText = "UART1_Rxed: ";
			debugTextLength = strlen(debugText);
			messageLength = strlen((char*)RxBuffer);
			if (debugTextLength + messageLength < RX_BUFSIZE) {
				memmove(RxBuffer + debugTextLength, RxBuffer, messageLength + 1); // +1 for null terminator
				memcpy(RxBuffer, debugText, debugTextLength);
				uart->txDataSize = (uint16_t)(messageLength + debugTextLength + 1);
				UART_Transmit_WithState(&uart2);
			} else {
				printf("UART1 Reception: Not enough space!\r\n");
			}
			break;
		case 2:
			debugText = "UART2_Rxed: ";
			debugTextLength = strlen(debugText);
			messageLength = strlen((char*)RxBuffer);
			if (debugTextLength + messageLength < RX_BUFSIZE) {
				memmove(RxBuffer + debugTextLength, RxBuffer, messageLength + 1); // +1 for null terminator
				memcpy(RxBuffer, debugText, debugTextLength);
				uart->txDataSize = (uint16_t)(messageLength + debugTextLength + 1);
				UART_Transmit_WithState(&uart1);
			} else {
				printf("UART2 Reception: Not enough space!\r\n");
			}
			break;
	}
}`

In main:
`		if(uart2.rxState == DATA_RECEIVED){
			printf("UART%s DMA Reception Completed with Size %u and with text %s: \r\n",uart2.Text, uart2.rxDataSize, RxBuffer);
			onIdle_UARTReceive(&uart2);
			//function_call specifically for lab3
//			on_ReceiveFrom_HTERM_To_UART2(&uart2);
			modMessage(&uart2);
//			UART_Transmit_WithState(&uart1);

		}
		if(uart1.rxState == DATA_RECEIVED){
			printf("UART%s DMA Reception Completed with Size %u and with text %s: \r\n",uart1.Text, uart1.rxDataSize, RxBuffer);
			onIdle_UARTReceive(&uart1);
			//function_call specifically for lab3
			modMessage(&uart1);
//			on_ReceiveFrom_UART2_To_UART1(&uart1);

//			UART_Transmit_WithState(&uart2);
		}`

Muslinmin, 2024-01-31:
```
554				uart->txDataSize = (uint16_t)(messageLength + debugTextLength + 1);
555				UART_Transmit_WithState(&uart2);
```

```
567				uart->txDataSize = (uint16_t)(messageLength + debugTextLength + 1);
568				UART_Transmit_WithState(&uart1);
```

Line 554 and 567... As mentioned earlier, the txDataSize should be called for uart2 and uart1 respectively.. It was mentioned earlier....

i.e.

`554				uart2.txDataSize = (uint16_t)(messageLength + debugTextLength + 1);`

`567				uart1.txDataSize = (uint16_t)(messageLength + debugTextLength + 1);`

As for calling the UART_TransmitWithState, just make sure its not called in the main loop. i.e. check if the previous implementation that calls the method in the while loop is commented/removed

@s2200931

Muslinmin, 2024-01-31:
Basically, in your switch case,

lets say UART1 receives the data. Inside the case statement, whatever data is received by UART1 is processed, and finally the final size of the buffer is computed. The txDataSize that should be updated is UART2, not UART1, because UART2 will be sending the buffer back to HTERM...

So it should be uart2.txDataSize rather than uart->txDataSize - cuz this one refers to txDataSize of UART1, NOT UART2 @s2200931

commit 42d8b49b (2024-01-30): Fix #18 

Fix #18
closed 2024-02-01 by Muslinmin